Integrated-Project-2-2021-UVic-UCC/rubik-s-cube #9: Mid-term presentation 22/03

davidcartiel, 2021-03-23:


marcsala27, 2021-03-25:
Here's our Mid Term Presentstion (Rubik's Cube Solver):
[Mid Term Presentation Rubik's Cube Solver.pdf](https://github.com/Integrated-Project-2-UVic-UCC/rubik-s-cube/files/6203942/Mid.Term.Presentation.Rubik.s.Cube.Solver.pdf)

Avoluciona, 2021-03-26:
Hello there! It's Aleix Villamor, from the Power glove device project, and I will give some feedback of your midterm presentation.

First of all, in general terms I enjoyed the presentation, it was fluid and the current status of the project was cleared up successfully. You have been self-critical during the exhibition, and that is so positive. It's good to know that you have reached the first objective, but I missed a little explanation about the other parts. The scanning and the solving algorithm parts seem so complexe and I expected some idea about the current status of those parts.

I think this project is so difficult, and you are so brave choosing this project. To reach the objectives of the project might be challenging, but it's seen that your cooperation is great, and its obvious that you can succeed because you work together since the first year in this career. I hope you are enjoying this subject and that you are learning a lot during the process.

See you in the final presentation and good luck!

SergiMuac, 2021-03-27:
Hi everyone!!

First of all congratulations on your job!!
Your presentation was very clear and well prepared. Your objectives seem to be well defined and work is well on track.
We only see one bad point, and it is related to face scanning. We didn't understand how do you think to scan the different faces of the cube and identify the colors. If you will use cameras we have thought that a good solution will be to create a secondary station where the Rubik's cube is scanned by two cameras placed in opposite corners diagonally and once it is scanned, the cube is placed in the solve station.

We would like to know what you think about our idea, and once more, congratulations on your job!!

Eye-Sphere Team

XaviMico17, 2021-03-28:
Dear Rubik's cube resolver team,

First, congratulations to you all, I enjoyed your exposition a lot.
There is very few I can recommend you about improving your presentation. It was very dynamic, coordinated, and interesting. 
As for to give you some advice, I would maybe hurry a little bit more at the beginning so at the end you do not have to shorten some parts to fit into the scheduled time, but just to say something to improve for the next, since there is not much more I can say.
I would also like to ask about the Rubik’s face detecting part. If I understood correctly, you are going to use a program to set manually the colour positions to give the motors the information about how to move. Are you going to program this code it yourselves? Also, would it be possible to use some cameras to track the colour instead of creating a program where you must give the input, or would it be too much of a burden?
I am excited to see how this project evolves and what ends up being because I think it is one of the most difficult of them all.
Have fun while developing it and good luck!!

Xavier Micó - Sun tracking device team

Guillem-Maza, 2021-03-28:
First of all, our team wants to congratulate the job you are doing.  
The presentation was very clear and fluid. For sure you had preparated the presentation well in order to explain correctly your project.  
All the members of our team think that it was very difficult the programming part of your project but in the presentation, you show us the way you are going to proceed and we believe that it is correct.  

One point that we don’t understood at all, is the part about how to send the start position of the cube to the algorithm code, and how you are going to get the information trough the photo done by a mobile phone or camera.  

Keep working and good luck! 
Covid device team.

davidcartiel, 2021-03-31:
> Hello there! It's Aleix Villamor, from the Power glove device project, and I will give some feedback of your midterm presentation.
>
> First of all, in general terms I enjoyed the presentation, it was [...]

Good afternoon and thank you for the message.
Right now we are working on the algorithm and also with the scanning part, so at that moment we couldn't explain so much about how we want to solve the cube, but first of all, we are going to try to do an approximation to the basic algorithm that everyone uses when they first try to solve the cube (@moritzhaecker can explain more about it). If we reach, and the algorithm works well, we will try to improve it to make it as fast as possible.

davidcartiel, 2021-03-31:
> Hi everyone!!
> 
> First of all congratulations on your job!!
> Your presentation was very clear and well prepared. Your objectives seem to be well defined and your work is well on track.
> We only see one bad point, and it is related to face scanning. We didn't understand how do you think to scan the different faces of the cube and identify the colours. If you will use cameras we have thought that a good solution will be to create a secondary station where the Rubik's cube is scanned by two cameras placed in opposite corners diagonally and once it is scanned, the cube is placed in the solve station.
> 
> We would like to know what you think about our idea, and once more, congratulations on your job!!
> 
> Eye-Sphere Team

Good afternoon all,
At this point of the project, we have not defined at all what kind of scanning we want to use exactly. Our options are two. The first one is like the one you propose, that is, create a place where to leave the cube and a camera that scans the colours of those 3 faces and then, change the cube manually to have the view of the other 3 faces. Once we have all the cube scanned we have to set it into the solver.
The other option is to set 2 cameras in diagonal attached to the solver, just have one station, to have the view of all 6 faces. If we do that, we have the problem that some of the parts, as we explained, will be invisible for our camera because of the axis, and we will have to do an extra algorithm to move the cube to find the colour of those parts. Hereinafter we will focus on all these parts to succeed.

davidcartiel, 2021-03-31:
> Dear Rubik's cube resolver team,
>
> First, congratulations to you all, I enjoyed your exposition a lot.
> There is very few I can recommend you about improving your presentation. It was very dynamic, [...]

Good afternoon and thank you for the recommendations,
At this time we have created a program where you give the configuration manually, to simulate the part of detecting the colours that we haven't done yet, and then, it knows from each part, what number corresponds to it, according with our numeration (this will be useful for the algorithm program). But our idea is to detect the colours automatically with the camera.
The algorithm code we are going to do it by ourselves with the knowledge of @moritzhaecker, who is the one who knows how to solve it faster.
About tracking the position of the cube, we think it's better to have an algorithm that, every move the robot has to do, it knows the new position, so we don't have to be scanning the cube each time and precessing the information (this algorithm is done). So, when we have the configuration of the colours, we will give them to the algorithm, and this, is going to solve the cube before moving it. Once it is solved, we know the movements that the robot has to do, and this information is the one we are going to send to the Arduino.

josepmaria-carbonell, 2021-04-06:
Good job guys!
The presentation was enthusiastic. We have seen that the color scanning part was the one which raises more qüestions.

However, the control of the movements for a speedy operation and the performance of the mechanics of the system is still an unknown. Some tests would be needed.

@Integrated-Project-2-UVic-UCC/rubik-s-cube

closed 2021-04-13 by marcsala27