Integrated-Project-2-2021-UVic-UCC/rubik-s-cube #23: Structure strains, deformations and displacements

marcsala27, 2021-04-13:
[Torre Definitiva Càlculs.pdf](https://github.com/Integrated-Project-2-UVic-UCC/rubik-s-cube/files/6303878/Torre.Definitiva.Calculs.pdf)

josepmaria-carbonell, 2021-04-14:
Hola @marcsala27, et poso les observacions en català:

   És una bona aproximació inicial. Però veient el pdf em surten diversos dubtes, que s'haurien d'afegir a aquests càlculs:

- No queda clar al final com apliques la força, és una pressió repartida a la superfície o una càrrega localitzada?
- No queda clar quines són les restriccions del problema on restringeixes els moviments?
- Altres possibles configuracions de càrrega i fixacions del motor en els cargols?
- Quin tipus d'elements s'utilitzen a l'anàlisi?
- Hauries de fer un anàlisi de convergència, fer els càlculs amb malles més fines i veure si convergeix al mateix resultat.
(per fer-ho t'has de fixar amb el punt on hi ha màxim desplaçament i fer un gràfic de convergència i en el punt on hi ha màxima tensió fer un altre gràfic de convergència).
- Per tant has de localitzar els punts on hi ha els màxims i comentar-ho. L'important no són els colors sinó els valors !
- Per tant també necesites posar número d'elements i de nodes de les malles que utilitzes pel càlcul (posar també alguna figura on es mostrin les diferents malles)
- Quan estiguis segur dels valors del càlcul, mires les tensions principals i compares amb les del material utilitzant un criteri de trencament dels que vam veure a Resistència de Materials.
- Donats els reultats, pots optimitzar l'estructura?

marcsala27, 2021-04-14:
Hola Josep Maria, he arreglat lo de la força aclarint que era una pressió repartida per la base marcada en vermell.
També he comentat les restriccions i fixacions de les cares de la torre.
I finalment he fet l'estudi amb una malla més petita i m'han donat els mateixos resultats numèrics i visuals pràcticament.
No he realitzat cap altre configuració de càrrega ja que la part que va collada no la collarem del tot fort ja que el motor encaixa just a la torre i esta ben fixat.
Merci.

closed 2021-06-03 by marcpujolar